3d-printing-canada/CR20-Pro-Firmware-3DPC #4: Mesh leveling does not work?

bbence84, 2020-04-04:
Is it just or the auto bed leveling adjusment does not work for some reason? I tried with both Bed Level on and off (which setting seems to disappear and also the value disappears), but when the first layer is being printed, the Z is not moving at all constantly... Is this normal?

jdeadman, 2020-04-09:
Have you done the bed level task?    you need to have that saved (under configuration) before it will start to use the mesh

bbence84, 2020-04-09:
> Have you done the bed level task? you need to have that saved (under configuration) before it will start to use the mesh

Yes. But when the print starts, it does not apply the mesh. Should I edit my gcode so that it applies the previously saved mesh? If yes, what? This is my start.gcode in S3D:

G28 ; home all axes
M117 Purging Extruder
G92 E0 ; reset extruder
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
G1 X0.1 Y20 Z0.3 F5000.0 ; move to start-line position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; draw 2nd line
; G1 E27 F3000 ; retract filament 3mm
G92 E0 ; reset extruder
; done purging extruder
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
M117 Printing...

jdeadman, 2020-04-09:
Ahh.   Now I see.   Yes you need to add the code

M420 S1 S2 ;Enable ABL using mesh and fade height. To your start gcode.

Poke255, 2020-05-09:
My son have a CR20 Pro and he has exactly the same issue ! Printer doesn't keep values stored by BL touch  :-/
unfortunately, the instruction M420 doesn't solve this problem... in addition, this line of code already exist in the gcode generate by Cura.
HELP !!! My son is desesperate

Hastur069, 2021-03-05:
Should be a "Z" not "S"

M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

AN63L, 2024-05-26:
For those of you still struggling with Bed Levelling issues on the CR-20 Pro in 2024, here is my working GCode that takes into account uneven beds: 

```
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home
M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
```